sh123/codec2_talkie #1: M17 support

mobilinkd, 2020-12-09:
Hi -- I am working with the [M17 project](https://m17project.org/).  M17 is based around Codec2 -- it's an open alternative to all the AMBE-based VHF/UHF digital modes.  My goal is to get an M17 transceiver working using an FM radio, Mobilinkd TNC3 and Android app.  I was about to embark on the process of writing what you have already written.  I have the RX side working in the TNC3 and expect have TX done this week.  I am still working on how to best implement the KISS interface for this as we need to send & receive link setup information.

I am reaching out to see if you would be interested in accepting PRs for this.  You are welcome to join us on IRC to see what we are all working on. https://m17project.org/irc/#

If you would like to test this with a Mobilinkd TNC3, please let me know.  I'd be happy to send one out to you.

sh123, 2020-12-09:
Hello, KISS has 0x06 command code (SetHardware), which could be used for modem configuration from the client. Also, commands from 0x07 up to 0xfe are not in use. Currently, app only sends `0xc0` 0x02 0xff 0xc0` to set `P` to maximum value thus disabling CSMA for real time speech. Was thinking about extending command codes (and maybe add support for different radio module profiles), so that client application can change frequency and modulation parameters.

mobilinkd, 2020-12-09:
Hi,
CSMA is disabled by setting full duplex.  It tells the modem that the receive channel does not indicate the state of the transmit channel.  Setting PPersist to max value is not specified to disable CSMA. The modem is still required to wait for a clear channel before PPersist is even checked.

The command is encoded in the lower nibble of the first byte, the type indicator.  The upper nibble is used to identify a specific modem.  That is typically 0.  However, it could also be used to indicate the TNC mode.

I will probably start with a fork and we can see what make sense to merge back.  I do need higher level protocol support for constructing and decoding the link setup frame.

Rob WX9O

sh123, 2020-12-09:
Sure, just make fork and move on. I just tried to keep app as simple as possible to mimic causal FM transceiver / walkie-talkie non-reliable behavior without protocol overhead and link setup to run at low bit rate.

closed 2021-01-15 by sh123