sh123/codec2_talkie #2: Serial connection

oh3gai, 2021-01-06:
Hi! 

Great app! I was wondering is it possible to use this app somehow without kiss protocol capsulation? I was thinking to try out ie. HC-02 serial rxtx ? 

This would make possible ultra low cost DV walkie talkie.

Thanks again for you great work!

Br, Jarno / OH3GAI

sh123, 2021-01-06:
That would be interesting, can add some checkbox to enable/disable streaming without KISS packetization.

oh3gai, 2021-01-06:
Hi!

Thanks for answer! Button for enable / disable kiss capsulation would be great. I was thinking to use below setup and avoid to playing with arduino. Keep the setup as simple as possible. OTG and HC12 Usb serial module.

[image]

Later on I will test app with Lora32u4 modules when they arrive from china.

br, Jarno

sh123, 2021-01-06:
I do not have HC-12, but it may not work out of the box with the app, kiss might not be the problem, default baud is set to 115200 in the application, HC-12 requires AT commands to change 9600 default baud rate. I have some ideas to add settings with required fields or radio module profiles to enable frequency changes, power and so on for different modules.

oh3gai, 2021-01-06:
Hi!

Sounds really good If you have time in some point to add and make changes. I can try to verify that. 

I have those HC12 modules and HC12 Usb adapters lying around from my greenhouse project. 

Br, Jarno

sh123, 2021-01-06:
Maybe, you can also try to configure it for 115200 baud rate if it does not work out of the box as described at https://picaxeforum.co.uk/threads/configuring-hc-12-usb-module-slightly-ot.30373/

oh3gai, 2021-01-07:
Good morning!

I was able to set up 115200,8,n,1, Programs startup nicely with my samsung A8(2018). 

I monitored transmission another hc12usb via putty serial. Below reception was done when counting numbers backwards. Unfortunately I can't verify the audio yet. The program wont start in my old honor (android 4). I need to get another phone. I'll report these results later. 

[image]

Br, Jarno

sh123, 2021-01-07:
Yes, it needs Android 6, also had old second phone, but managed to flash it with Cyanogenmod newer Android version as manufacturer does not publish new versions, maybe try to flash it with [Cyanogen Android 6.0.1](https://forum.xda-developers.com/t/rom-6-0-1-cyanogenmod-13_20-04-2016-for-huawei-honor-3c-h30-u10.3363980/)

oh3gai, 2021-01-07:
Hi! 

I can verify that audio works path works between two otg&HC12usb dongles with parameters 115200,8,1. 

Distance is really poor with speed of 115200. :)

Great app!

br, Jarno

sh123, 2021-01-07:
Good to hear, just set lower speed for modem, but keep 115200,8,1 serial speed, this will give greater distance. With 900 bps LoRa  and 450 bps codec2 mode I was able to get around 5 km when light of sight is not obstructed.

oh3gai, 2021-01-07:
I need to google a bit more about HC12 usb adapter. 

When I change the rf path way speed the speed between pc and hc12 is changed also.  

With 1200-2400baud bandwidht seems to be reasonable 30khz and with 115200 around 400khz.
[image]

[image]

br, Jarno

sh123, 2021-01-07:
It will take some time to implement proper settings, for now just published apk-s for different serial baud rates https://github.com/sh123/codec2_talkie/releases/tag/0.13

oh3gai, 2021-01-08:
Thanks for solving this.

I will make some more codec2 mode and range test near future with HC12 .

br, Jarno

closed 2021-01-14 by oh3gai

AvtechScientific, 2021-07-22:
> Hi!
>
> Thanks for answer! Button for enable / disable kiss capsulation would be great. I was thinking to use below setup and avoid to playing with arduino. Keep the setup as simple as possible. OTG [...]

Hello,

nice to see that somebody has thought about using HC12 connected via USB cable to  the Android phone before me. Were you able to use such a setup successfully? 

What I want to do is just to enable simple low-speed/long range communication  (like send a short string or and integer once in a while) between an Arduino Nano and an Android phone via two HC12 radios... As simple as Nano sends "Hello world!" and Android shows it on its screen...

1. I have bare HC12 - how should I solder an USB cable to it so Android can receive data this way?
2. Which Android app should I use to see such simple messages arriving from the Nano?

Thank you!

sh123, 2021-07-22:
@AvtechScientific , I believe, you need OTG cable and some USB serial applicaiton, like https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal

AvtechScientific, 2021-07-22:
> @AvtechScientific , I believe, you need OTG cable and some USB serial applicaiton, like https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal

Thank you very much for your fast response! Just - how should I solder the HC-12 to the OTG Usb cable? HC-12 that I have look [like this](https://www.aliexpress.com/item/4001052642489.html?spm=a2g0o.search0304.0.0.46017b23RUB61Y&algo_pvid=298f4b3a-fa3a-403f-831e-a2595cade123&algo_exp_id=298f4b3a-fa3a-403f-831e-a2595cade123-0) and not like as in the picture above... HC-12 has 5 pins/holes...

AvtechScientific, 2021-07-22:
> @AvtechScientific , I believe, you need OTG cable and some USB serial application, like https://play.google.com/store/apps/details?id=de.kai_morich.serial_usb_terminal

By the way - can I play an alert or notification sound on such USB serial applications once a message from Arduino Nano arrives?

sh123, 2021-07-22:
@AvtechScientific , the module on the picture is not HC-12, but HC-12-USB, raw HC-12 cannot be used through USB, it uses SPI bus, to use it with the phone you need to use it with some MCU, like Arduino and then connect to the phone via USB OTG.

awatchar, 2023-10-31:
[image]
[image]

I try with Ebyte E22-400T22U also work very well.
Need to adjust air rate a little bit higher than default with CODEC2_450. 

It can work with higher Codec2 bit rate but more air rate need to be used.