sh123/codec2_talkie #4: PTT button support.

oh3gai, 2021-01-08:
Hi!

I have inrico tm7 (android 6.0) network mobile radio which have mic and real ptt button. It would be  great to see support PTT. 

It would make full featured real Digital Voice radio for it with quite slight modifications. Just add radio modem of your choice with otg cable.

Codec2Talkia apk was easy to install to inrico tm7. Just copied apk over usb cable and everything works out of the box via HC12 radio module.
 
[image]

Below is picture of Inrico TM7 keycode event 230 when PTT is pressed and released
[image]

You have created great app with lot of possibilities!!

Br, Jarno

sh123, 2021-01-08:
Wow, that is great use for network radio, check https://github.com/sh123/codec2_talkie/releases/tag/0.14 release, I've mapped button with 230 key code (KEYCODE_TV_DATA SERVICE) to PTT UI button.

oh3gai, 2021-01-08:
Hi!

Awesome work. Works like charm..

https://user-images.githubusercontent.com/37511358/103994727-1393c280-51a0-11eb-93f9-ed1ee8a8466b.mp4

I will be back with grazy ideas..

Thanks a lot!

Br, Jarno

closed 2021-01-08 by sh123