sh123/codec2_talkie #5: Quesition: Decode codec2talkie with c2dec

oh3gai, 2021-01-13:
Hi!

Sorry to bother again..

I've been trying to decode codec2talkie transmission(HC12) via ubuntu c2dec, but without success.  Audio is not understandable.

"sudo cat /dev/ttyUSB0 | c2dec 700 - - |  play -t raw -r 8000 -e signed-integer -b 16 -c 1 -"

[image]
  ###

Have you tried that? Could it be the kiss encapsulation? I tried to understand the code but unfortunately my competence was not enough.. :)

Thanks for you're help!

Br, Jarno

sh123, 2021-01-13:
Hey, yes, that's because of KISS, you need to extract audio frames from KISS packets first before playback.

oh3gai, 2021-01-14:
Hi!

Thanks for info. I was able to install kiss module to python, but needs figure out how to set print format show that it just passes all the info to stdout without adding those new lines etc. That's propably preventing the piping.

[image]

Thanks for help!

br, Jarno

sh123, 2021-01-14:
Probably something like this will work out
`
import sys
for b in byte_array:
    sys.stdout.write(chr(b))
`

oh3gai, 2021-01-14:
Hi!

Kiss module started  to print stdout with 

sys.stdout.buffer.write(x)

[image]

I'll continue investigations after I get back from work.

Br, Jarno

closed 2021-01-14 by oh3gai