sh123/codec2_talkie #8: Feature request: Add support for SX1262, SX1268 and SX1280 LoRa chips from Semtech

happytm, 2021-07-30:
Thank you for your wonderful work on codec2.

Can you please add support for newer LoRa chips mentioned above. There is a library with lot of examples which support all of these chips and old chips too like SX1276 and SX1278:

https://github.com/StuartsProjects/SX12XX-LoRa

Thanks.

sh123, 2021-07-30:
@happytm , this project does not operate on LoRa chips directly, it is Android application, which operates over bluetooth/usb by using KISS protocol, it requires a modem, which operates on the given chip.

closed 2021-07-30 by sh123

happytm, 2021-07-30:
I understand this is an android app but I was asking support for newer Lora chips on companion sketch you have for https://github.com/sh123/esp32_loraprs

I just wanted to experiment with it as a walkie-talkie using codec2 compression.

Thanks.

sh123, 2021-07-30:
Currently, it uses https://github.com/sandeepmistry/arduino-LoRa library, you can ask its owner to add support for newer chips, currently it supports Semtech SX1276/77/78/79 chips, switching to another library requires quite a lot of work and I do not even have these new ships to test.

happytm, 2021-07-30:
OK.Thanks.