sh123/codec2_talkie #19: Mic gain

madbird99, 2021-10-30:
Testing the app between two phones, one with Android 5.0 (legacy) and the other with Android 10 there is a huge difference in the mic gain. In the legacy app the mic gain seems very low. I did not find how to adjust it.

sh123, 2021-10-30:
@madbird99 , seems to be phone related, I have older android phone with higher gain mic, new one mic gain was louder, mic was full of dust, after cleaning seems to be better.

madbird99, 2021-10-30:
Just for reference, on the same phone the Echolink app works perfectly without increasing the mic gain. (0db).

sh123, 2021-10-30:
I'll add ability to specify different audio source from settings, currently it uses VOICE_COMMUNICATION, so it performs AGC and noise cancellation, maybe it causes this low gain in your case.

sh123, 2021-10-30:
Also, set "Play audio through the speaker" in settings, maybe it plays through the ear phone, so it sounds quiet.

madbird99, 2021-10-30:
> Also, set "Play audio through the speaker" in settings, maybe it plays through the ear phone, so it sounds quiet.

I just checked, it's ok

sh123, 2021-10-31:
https://github.com/sh123/codec2_talkie/releases/tag/1.20

added option to settings to enable VOICE_COMMUNICATION, which enables noise cancellation and echo supression, by default it will be switched off, so mic gain will be higher

closed 2021-10-31 by sh123