sh123/codec2_talkie #20: Wrong android API requirements

dtrenin7, 2021-11-15:
You defines "Android 7.0 (API 24) or higher"  in system requirements, but this code requires API 28 (or higher) really.
This makes old phones unusable.....

sh123, 2021-11-15:
Usually using 3 devices to test:

- codec2talkie-release.apk works fine on Android 7 API level 24 and new Android 11
- codec2talkie-release-legacy.apk works fine on Android 5.1.1 API level 22

not sure what is your problem without more details

closed 2021-11-16 by sh123

dtrenin7, 2021-11-16:
On API 28 build is OK.
When i switched to API 24 multiple resource linkage errors comes...

[http://joxi.ru/DmBoO5LUqY4jR2](http://joxi.ru/DmBoO5LUqY4jR2)

First is "AAPT: error: style attribute 'android:attr/dialogCornerRadius' not found."
As i knew, in another issue you mentioned switching to API 28+ as a solution.

sh123, 2021-11-16:
you need to use another branch for older models if building yourself, there is legacy branch

dtrenin7, 2021-11-16:
Thankx you!
Builded for 21 SDK after that)

but your config contains hardcode paths - no good....
remove "/usr/bin/cmake" with "cmake" (relative path) in build.gradle files.....
path to cmake may vary on different systems