sh123/codec2_talkie #21: Dual Channel / Repeater Mode

mrpatrick1991, 2022-03-09:
It would be really cool if we could specify different channels / spreading factors for transmit and receive. If these could be changed on the LoRa TNC using KISS extensions when the PTT button is pressed, you could create a low-power repeater using two TNC's wired together over their serial ports.

A friend and I are going to fly some LoRa TNC's on high altitude balloons, and would love this feature.

sh123, 2022-03-10:
Do you mean split mode? I can add it so that you will be able to specify separate RX and TX frequencies, also need to add support into the modem.

mrpatrick1991, 2022-03-10:
Yes exactly, different frequencies would work well. Eventually we want to send some long-distance solar powered superpressure balloons with LoRa repeaters on them which people with the LoRa ESP32 modem can use!

sh123, 2022-03-10:
Balloon with repeater sounds interesting, will add this split into modem and this app once have free time.

mrpatrick1991, 2022-03-10:
When we try it, I'll post back with recordings and data from the flight. We can cut down our payload weight by about a factor of 10 compared to using 2 analog FM HT's.

sh123, 2022-03-11:
@mrpatrick1991 , added support into the modem, you can set different frequencies for RX and TX in config and it will toggle to TX on transmit and back to RX on receive. Did not test.

mrpatrick1991, 2022-03-11:
Wow, that’s awesome! I’ll test a repeater setup and let you know how it works. I think we might be able to use the orthogonality of the LoRa spreading factors to avoid de-sense on the same band - I might try making a PR with the option to change spreading factors between TX and RX as well.

happytm, 2022-05-04:
@mrpatrick1991 I am curious to know if repeater setup worked for you or not ?

Thanks.

sh123, 2022-06-20:
@happytm , @mrpatrick1991 , any updates on it, does it work, can I close it?

mrpatrick1991, 2022-06-20:
I’ll try it this week and write back!

happytm, 2022-06-20:
Thank you @sh123 and @mrpatrick1991 .

happytm, 2022-08-03:
@mrpatrick1991 Any update on your test ?

Thanks.

closed 2022-08-13 by sh123