sh123/codec2_talkie #23: please increase versionCode

IzzySoft, 2022-06-19:
Please remember to increase `versionCode` before you make the next release: it is still [set to `1`](https://github.com/sh123/codec2_talkie/blob/master/codec2talkie/build.gradle#L13) despite of many versions already having been released. For background: Android uses `versionCode` to tell which APK is newer and offer it to the user for updating. If it wasn't changed, Android assumes this is the very same version, making it more difficult to update (and especially not sending update notifications). Thanks!

PS: I've just added your app to my F-Droid repository (it will [become available here](https://apt.izzysoft.de/fdroid/index/apk/com.radio.codec2talkie) with the next sync around 6 pm UTC). New versions will be fetched within 24h of your tagging a new release and attaching the APK to it. Users of my repo will receive update notifications for new versions – provided you have increased `versionCode` :wink:

If you like to link to your app's new presence, be welcome to [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it) for visualization :smiley:

sh123, 2022-06-19:
Increased in master, will be correct with the next release.

closed 2022-06-19 by sh123

IzzySoft, 2022-06-19:
Wonderful, thanks!