sh123/codec2_talkie #46: Feature request: Target location direction indicator

happytm, 2022-09-28:
I have been following your great project for some time now.

I would like to request a feature which could be useful for pointing directional antenna to target specific location (repeater antenna or another smartphone).

I came across an android app that do same and it is linked below. 

https://play.google.com/store/apps/details?id=ea4eoz.HamGPS&hl=en_US&gl=US

Thanks.

sh123, 2022-09-29:
I have HamGPS installed, too. To achieve similar result you can open APRS map and select "Rotate map with compass", this way you can point your phone and directional antenna to the desired direction.

happytm, 2022-09-29:
Do I have to be ham operator to use this feature?

Thanks for your reply.

sh123, 2022-09-29:
It just shows map centering at your current position if GPS is enabled, to send data you need to explicitly start beaconing.