sh123/codec2_talkie #47: Add Google Lyra Support

faydr, 2022-11-02:
Google's made some impressive progress with their AI-based Lyra vocoder. 

[Link to Google Lyra Github](https://github.com/google/lyra) 

Integrating this codec into codec2 walkie talkie would make for an amazing walkie-talkie app in terms of voice quality!

sh123, 2022-12-01:
@faydr , good idea, but their minimum bit rate is 3kbps, which is going to work only on 125khz LoRA channels and with Codec2 can fit into 25 kHz UHF channels at worse quality, but still readable.

Tfjjjastro, 2023-12-05:
I think, Lyra would be an excellent option than opus because it has capability of running on low end devices as well as it need less processing, it has capability to run on older Android devices so maximum support to all devices and i have listen the Lyra samples, it's actually giving mind boggling results at being 3.2kbps maybe so please please implement this on your app it would be a wound to this app, much better than codec 2 and other rest audio codecs, it has also much bigger support

closed 2025-10-29 by sh123