sh123/codec2_talkie #49: App crash then try choose aprs symbol

US1GHQ, 2023-05-07:
Good day.
First thanks for this great aplication and firmware for LoRa APRS modem.
Second Then i trying choose diferent APRS symbol aplication is crash. i am using Android 12 and Poco X3 Pro, version of codec2_talkie, latest from your release page, other functions working perfect, exept coosing aprs symbol.
Thanks and sorry for my English =)

sh123, 2023-05-08:
Hello, try with new 1.54 release, include logcat output if possible at the time of the crash if it will happen again.

US1GHQ, 2023-05-08:
Good day, i am trying 1.54 and it is still crashing.
I am trying include LogCat output, maybe it is help =)
[codec2_log.log](https://github.com/sh123/codec2_talkie/files/11419842/codec2_log.log)

US1GHQ, 2023-05-08:
Now it s all fine all working, thank you very much.
Have a nice day;-)

closed 2023-05-08 by US1GHQ