sh123/codec2_talkie #51: Suspected APRS-IS data loopback issue

hessu, 2023-06-29:
Hi,

Yesterday someone noticed that an APRS-IS client has kind of heard roughly all the other stations around the world on APRS-IS and retransmitted the packets back. Take a look at the rather lengthy "heard directly by" list of this iGate, and hover your mouse cursor on top of the "longest at" timestamps to see the respective packets:

  https://aprs.fi/info/a/EA3HCP-7

Apparently the igate has sent out two packets back-to-back with a "<0xd" string between the two packets. There are also a lot of brandmeister-originated DMR packets with a ,qAS,EA3HCP-7 construct. I sent a mail to the callsign owner asking which software it was running at the time, and he reported he was running the codec2_talkie app on his Android device at the time.

I didn't really step through the code properly, and I'm not much of a java programmer, but I wonder if AprsIs.java receive() will be called with received APRS-IS data, and and will it call sendData which, if _isSelfEnabled, will put the received data back to the APRS-IS txQueue ? This is just a rough guess based on a cursory look.

https://github.com/sh123/codec2_talkie/blob/master/codec2talkie/src/main/java/com/radio/codec2talkie/protocol/AprsIs.java#L124

Perhaps during the implementation of #40 such a bug crept in?

sh123, 2023-06-29:
@hessu , sorry about that, I think there is a typo, it should call _childProtocol.sendData, but not sendData, I'm about to fix it and make new release.

sh123, 2023-06-29:
Fixed in 1.56 version

closed 2023-06-29 by sh123

hessu, 2023-06-29:
Wait, there's more, I'm afraid. Stay tuned for details.