sh123/codec2_talkie #62: Can app be more like whatsApp or any similar app can be made

Tfjjjastro, 2023-12-11:
Can this or similar app be made but it should use this app Protocol for packet exchanging, I mean an app like whatsApp where you can text, Send file with using xmodem file transfer or any better option, Encryption, own identify, I mean this could be more awesome, also an app which maybe more simple than this in many ways so that normal user could also use like we can write good, better, extreme, poor, normal other than writing different bitrate by own, also many other option for more scope in being more simple way so that normal person can actually use.

sh123, 2023-12-11:
Narrowband radio channels do not have enough bandwidth for all that internet stuff. Need to be some sort of spacex to give that to you.

Tfjjjastro, 2023-12-23:
Can i get your personal mail?

sh123, 2023-12-25:
@Tfjjjastro , hi, what do you want to ask, why do you need contact in private?

Tfjjjastro, 2023-12-25:
I need your help in being making app for my product.

sh123, 2023-12-25:
I'm not doing commercial projects, sorry

Tfjjjastro, 2023-12-25:
But it's a start up, I think, you can help

Tfjjjastro, 2023-12-25:
Please don't say no

Tfjjjastro, 2023-12-25:
Would you be help in being public work on GitHub?

closed 2023-12-25 by Tfjjjastro