sh123/codec2_talkie #63: Little bit advancement in app

Tfjjjastro, 2023-12-29:
[image]
Please give a side option in app for chatting with kiss protocol and an option for photo, pdf and audio message sending and receiving , I think this could possible with kiss protocol i mean this could really be helpful in many situations, I. Hope you may like this idea too.

sh123, 2023-12-30:
There are two problems. First is that APRS protocol does not support file transfer (and APRS UI frames are not reliable to do that) and second is the size of the image versus channel capacity. For example, 6 MB typical photo made on the phone will be transferring for about 10 minutes over 9.6 kbps channel. Perhaps, you can run on 868/915 MHz with wider bandwidth, but there are some ISM band plan limitations for how long you can occupy the channel.

Tfjjjastro, 2023-12-30:
That's true. I am actually working on a transceiver which uses other advance modulation techniques to transfer many different types of data over longer range for better communication. TCP/IP would take long time and complexity and efforts there are other other kinda file transferring protocol like xmodem and ymodem documention on well manner for Android and ios.

closed 2024-01-02 by Tfjjjastro