sh123/codec2_talkie #65: Peer to peer walkie talkie

Tfjjjastro, 2024-01-09:
Like what is there now in app, it's nice, but can we include an identification system for more privacy and encryption too, like in app, there should be an option for making user profile, or randomly assign identity to every app that could be use in verifying the user and only to talk to those whom we are calling too

sh123, 2024-01-09:
I doubt it is properly doable without central server on low bandwidth radio channels.

Tfjjjastro, 2024-01-09:
if many people using same frequency of lora module and this could actually be mess up like many people will started listening our conversation, to be more private the conversation of this identity system would be there so it will be awesome

Tfjjjastro, 2024-01-09:
Think like give an unique identity to each phone which using the app, first a packet would be send, if the same recipient same the packet who we want talk with, a secure connection would square and audio would be sent trap in encryption and a decrypt key would have recept so read only and only no one else

closed 2024-01-13 by Tfjjjastro