sh123/codec2_talkie #66: Problem with compiling to apk on ubuntu 22.04

AlonSpinner, 2024-01-17:
I have a custom modem that I would like to use with custom data transfer protocol through the usb, and hence I am trying to compile the project and build an apk.
This is my first time touching any java project, and I am a complete noob XD.

### prerequisites
running on ubuntu 22.04, this was my setup:
```
sudo apt update
sudo apt install openjdk-11-jdk sdkmanager
```
following with:
```
sdkmanager --licenses #accepts all licenses
sudo update-alternatives --config java #---> choose java11
#exported ANDROID_SDK_ROOT to the root of the project
export ANDROID_SDK_ROOT=$(pwd) #---ensure correct folder
export JAVA_HOME=/usr/lib/jvm/java-11-openjdk-amd64
```
### Compiling to apk
`./gradlew assemble` OR `./gradlew build`
This didnt work. please see below what I did to "fix" problems, and where I ended up.

#### changes

in ./build.gradle I changed classpath from 3.5.0 to 4.1.1 to avoid:
```
Execution failed for task ':libcodec2-android:stripDebugDebugSymbols'.
> No toolchains found in the NDK toolchains folder for ABI with prefix: arm-linux-androideabi
```

in ./codec2talkie/build.gradle I changed the compileSdkVersion to 31 from 30 to avoid:
```
* What went wrong:
Execution failed for task ':codec2talkie:checkDebugAarMetadata'.
> Multiple task action failures occurred:
   > A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
      > The minCompileSdk (31) specified in a
        dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
        is greater than this module's compileSdkVersion (android-30).
        Dependency: androidx.preference:preference:1.2.0.
        AAR metadata file: /home/alon/.gradle/caches/transforms-2/files-2.1/ac9da05f86a847f6bc2d750040f5f454/preference-1.2.0/META-INF/com/android/build/gradle/aar-metadata.properties.
   > A failure occurred while executing com.android.build.gradle.internal.tasks.CheckAarMetadataWorkAction
      > The minCompileSdk (31) specified in a
        dependency's AAR metadata (META-INF/com/android/build/gradle/aar-metadata.properties)
        is greater than this module's compileSdkVersion (android-30).
        Dependency: androidx.lifecycle:lifecycle-livedata:2.5.1.
        AAR metadata file: /home/alon/.gradle/caches/transforms-2/files-2.1/d3cf9c06a879226a8ed322c5fd6bb964/lifecycle-livedata-2.5.1/META-INF/com/android/build/gradle/aar-metadata.properties.
        ...

```

after which I got the following error:
```
* What went wrong:
Execution failed for task ':libcodec2-android:mergeDebugNativeLibs'.
> A failure occurred while executing com.android.build.gradle.internal.tasks.Workers$ActionFacade
   > More than one file was found with OS independent path 'lib/x86_64/libcodec2.so'. If you are using jniLibs and CMake IMPORTED targets, see https://developer.android.com/studio/preview/features#automatic_packaging_of_prebuilt_dependencies_used_by_cmake
```

searching for where all the libcodec2.so with `find . -name libcodec2.so yielded`:
```
./libcodec2-android/build/codec2_build_android_x86_64/src/libcodec2.so
./libcodec2-android/build/codec2_build_android_x86/src/libcodec2.so
./libcodec2-android/build/codec2_build_android_arm64-v8a/src/libcodec2.so
./libcodec2-android/build/intermediates/cmake/debug/obj/arm64-v8a/libcodec2.so
./libcodec2-android/build/intermediates/cmake/debug/obj/armeabi-v7a/libcodec2.so
./libcodec2-android/build/intermediates/cmake/debug/obj/x86_64/libcodec2.so
./libcodec2-android/build/intermediates/cmake/debug/obj/x86/libcodec2.so
./libcodec2-android/build/intermediates/merged_jni_libs/debug/out/arm64-v8a/libcodec2.so
./libcodec2-android/build/intermediates/merged_jni_libs/debug/out/armeabi-v7a/libcodec2.so
./libcodec2-android/build/intermediates/merged_jni_libs/debug/out/x86_64/libcodec2.so
./libcodec2-android/build/intermediates/merged_jni_libs/debug/out/x86/libcodec2.so
./libcodec2-android/build/codec2_build_linux/src/libcodec2.so
./libcodec2-android/build/imported-lib/arm64-v8a/libcodec2.so
./libcodec2-android/build/imported-lib/armeabi-v7a/libcodec2.so
./libcodec2-android/build/imported-lib/x86_64/libcodec2.so
./libcodec2-android/build/imported-lib/x86/libcodec2.so
./libcodec2-android/build/codec2_build_android_armeabi-v7a/src/libcodec2.so
```
Unfortunately, now I am stuck, because from what I see, libcodec2.so was compiled for each architecture correctly, with no overlaping

AlonSpinner, 2024-01-17:
SOLVED?

I added the following to  libcodec2-android/build.gradle
```
android {
    packagingOptions {
        pickFirst 'lib/armeabi-v7a/libcodec2.so'
        pickFirst 'lib/arm64-v8a/libcodec2.so'
        pickFirst 'lib/x86/libcodec2.so'
        pickFirst 'lib/x86_64/libcodec2.so'
    }
}
```

I added the following to  libopus-android/build.gradle
```
android {
    packagingOptions {
        pickFirst 'lib/armeabi-v7a/libopus.so'
        pickFirst 'lib/arm64-v8a/libopus.so'
        pickFirst 'lib/x86/libopus.so'
        pickFirst 'lib/x86_64/libopus.so'
    }
}
```

any better suggestions?

sh123, 2024-02-18:
```
Execution failed for task ':libcodec2-android:stripDebugDebugSymbols'.
> No toolchains found in the NDK toolchains folder for ABI with prefix: arm-linux-androideabi
```

Hi, you need to use NDK 22 version, adding this line to build.gradle may help..
```
    android.ndkVersion "22.0.6917172"
```

closed 2024-02-18 by sh123