sh123/codec2_talkie #67: Superframe length and transmission bitrate do not match settings

TrokhymButko, 2024-02-13:
When choosing codec2 450bps, If I set superframe = 30 bytes, then in reality the transmission occurs in packets of 27 bytes. Since a regular frame is 3 bytes, I can assume that probably one frame is lost, but this is just my guess. These 27 byte superframes follow with a period of 360 milliseconds, which gives a bit rate of 600bps (but not 450). Are there any explanations?

sh123, 2024-02-18:
Hi, maximum means that you need to have this value defined as 31 to get 30 bytes length. 450 bps on air is not achievable, because lora adds preamble and other technical data to the packet, also modem has some delay between packets as it it is not streaming.

closed 2024-02-18 by sh123

TrokhymButko, 2024-02-20:
You don’t understand, instead of 450 bits per second there are 600 bits per second of pure audio codec data. Of course, I did not include the service data of the lora and KISS protocols here.

sh123, 2024-02-20:
App does not have to send at 450 bps, modem needs to queue and then transmit and desired bit rate.