sh123/codec2_talkie #68: LORA Uart modules support

Tfjjjastro, 2024-02-18:
Can you please add an additional feature in the app like those uart modules, I do have, it uses first 3 bytes of sending data in something as 2 bytes of data as address and 3rd byte as channel afterwards following data transmit to the particular address module, please give an additional option in setting for typing 2 byte address in hex code as (A1CF) and 1 byte for channel in hex code as (01) finally a place where user can. Input 3 bytes hex code as A2CF01 which includes all 3 bytes as header of streaming data.  Please introduce this option because in market, all lora uart module work like this. 
So that other users could also benefit this

sh123, 2024-02-18:
Hi, will try to find time to add it, do you have link to the specification for the modem?

Tfjjjastro, 2024-02-18:
[image]

This is the manual of the module

Tfjjjastro, 2024-02-18:
[image]

Page no. 15 has something relevant to this

sh123, 2024-02-18:
Module has 3 modes, to modify registers it needs to be in 3rd mode, which is set by changing M0 and M1, app does not have access to those, I'm not sure it makes sense to add, you need to pre-configure modem before usage and then app will transfer data in mode 0

[image]

Tfjjjastro, 2024-02-18:
Don't take care of modes, it can easily be control through switches, to high and low M0 and M1

sh123, 2024-02-18:
App needs some dedicated dialog for that, so that user can send configuration commands and manually change switches back before transmission.

Tfjjjastro, 2024-02-18:
The configuration mode is for setting up address and other things of modules and i don't want to change any setting via serial, it's just I need a setting option in app for adding header to the streaming audio data of codec 2 app to send the data to particular address, module automatically would recognize first 3 bytes as sender address and further back data as true data which would display on sender serial. Also header should use for each and every break like non continuous packet should have a address too so adding header is required.

sh123, 2024-02-19:
Do you mean WOR Mode 1? It says that preamble is added automatically or am I missing something?

[image]

Tfjjjastro, 2024-02-19:
First of all, we aren't changing any module configuration, second thing, we are on mode 1 of the module, third thing is like each and every module has it's own unique address, unique address has total 3 bytes in hex format compromises total 6 hex digits and whenever we send serial data to module, serial data should have a header which would compromises of 6 hex digits, this header helps to sent data to particular address, for this we don't need to high or low any pins of module, it's just module should be on mode 1 in which it transmits data, nothing else module automatic reads serial data upfront 3 bytes as that unique address where it needs to send the data. So whenever we sent data to module through serial we need add a header, header+data. If streaming data to serial has breaks then we again have to add header for each and every segment

Tfjjjastro, 2024-02-19:
What's what I am thinking in my mind, I am 21 years old and i am from India, New delhi
I don't have any budget, I have no team, I just trying to fund this thing through Kickstarter and indegogo. 
Do you think it's a good idea. 
[image]

[image]

Introducing RAIO, your ultimate communication companion! This pocket-sized device, with a sleek 15mm thickness, connects people up to 10km away through text and voice with a bandwidth of up to 36kbps. It effortlessly overcomes obstacles in the 850-950MHz band, ensuring seamless communication globally.

RAIO offers free license communication, empowering users with file transfer capabilities at 2.4GHz band with bandwidth upto 75 to 100kbps in being 4km distance, providing a broader space for interaction. Its dual-band support (850-950MHz and 2.4GHz) and 1-watt power limit make it versatile for any location, while the durable body ensures resilience in any environment, with a working temperature range of -10 to 80°C.

Boasting a lithium-ion battery, RAIO ensures extended runtime with minimal weight, running up to 3 days and 7 days on standby. The device features two dials for power and frequency adjustments for each band, giving users fine control. Additional ports for direct battery input, serial port for RAIO TAT peripherals which gives you more control and USB Type-C charging enhance versatility.

Pairing RAIO with your Android device via Bluetooth through the RAIO Experts App opens a world of possibilities. Make calls, send texts, and transfer files seamlessly. The app automatically adjusts wireless settings according to country regulations, ensuring hassle-free use worldwide. For radio enthusiasts, the app's developer mode allows advanced customization.

Join us on this groundbreaking journey – support RAIO, the device that transforms communication! Your backing will help bring this innovative product to life, connecting people like never before. Together, let's make RAIO a reality! #RAIOConnects #CommunicationRevolution

E220-900T30D taking as lora module for 850-930 mhz band and E34 2G4D27D for 2.4ghz, also we can enhance module performance and combine this module on single board from ebyte r&d department[image]

Tfjjjastro, 2024-02-19:
[image]

Tfjjjastro, 2024-02-19:
Ebyte haven't describe everything briefly but i have checked with Arduino by doing as instructions, it's actually working, so we have to add 3 byte header on streaming data.

sh123, 2024-02-19:
Try https://github.com/sh123/codec2_talkie/releases/tag/1.69, did not test myself as do not have this modem

closed 2024-02-19 by sh123

Tfjjjastro, 2024-02-19:
Thank you so much for the update but as I am using hc05 with uart lora and your update is limited to UART TNC, so my lora of tx-rx of hc05 and lora of rx-tx of hc05, wouldn't work here so please try to implement on Bluetooth too, it's a request