sh123/codec2_talkie #69: Request for prefix of Bluetooth

Tfjjjastro, 2024-02-19:
Thank you so much for the update but as I am using hc05 with uart lora and your update is limited to UART TNC, so my lora tx-rx of hc05 and lora rx-tx of hc05, wouldn't work here so please try to implement on Bluetooth too, it's a request

sh123, 2024-02-19:
Maybe it is time for you to implement it yourself and create pull request? I even cannot test your setup.

Tfjjjastro, 2024-02-19:
Okay I will let you the results with the recent update soon and try to contribute from my end but there is a problem with usb TNC, I am using redmi note 10 and the raspberry Pi pico with uart lora and i can connect to my device through usb serial terminal by kai-morchi but codec 2 app couldn't able to connect with pico, sometimes it's showing like it's connected but however it just disconnecting.

sh123, 2024-02-19:
https://github.com/sh123/codec2_talkie/wiki/Frequently-asked-questions

[image]

sh123, 2024-02-19:
Try https://github.com/sh123/codec2_talkie/releases/tag/1.70 with bluetooth, did not test, do not have your setup.

closed 2024-02-19 by sh123