sh123/codec2_talkie #70: Update 1.70 isn't working

Tfjjjastro, 2024-02-22:
I have checked 1.70 it's actually not working

Tfjjjastro, 2024-02-22:
With lora e220 900T30D

Tfjjjastro, 2024-02-22:
I do have connected that uart module with hc05, in order of tx-rx and rx-tx connections, I do have connected the hc05 with the app and 2nd uart transceiver with raspberry Pi pico to monitor the data on Arduino ide serial monitor, I also have installed Bluetooth serial monitor on phone and try to send a packed to my target address with header, it got sent so maybe the new function isn't working also if app even just write the address on Bluetooth then definitely I got something on serial monitor of 2nd end as I have checked on my previous tests that whenever I just type the hex address on each of 2 uart module then on 2nd one, it just show some gap Or something so i could understand that something has arrived but i don't think, app is properly even writing the hex address or maybe address in middle of packet, I don't know I am just guessing or maybe some other issue.

sh123, 2024-02-22:
Each transmitted packet is prefixed with correct 3 byte sequence when its enabled, I'm using app with KISS disabled, as you can see from the dump, first 3 bytes are COFFEE is the prefix for UART modem, then 9 3 byte Codec2 450 bps mode audio frames are following:
```
C0 FF EE 81 80 00 81 80 00 81 80 00 81 80 00 81 80 00 81 80 00 D8 1F C0 81 A0 00 2F 80 00 
```