sh123/codec2_talkie #72: ninja error:missing and no known rule to make it

BadDateAlex, 2024-06-14:
I was trying to make your project on my Android Sudio and I had this error. I don't know why there isn't a folder named imported-lib,did I miss sth? please help me, thank u.
`ninja: error: 'E:/Android/codec2_talkie-master/libcodec2-android/build/imported-lib/arm64-v8a/libcodec2.so', needed by 'E:/Android/codec2_talkie-master/libcodec2-android/build/intermediates/cxx/Debug/4gx1kb3u/obj/arm64-v8a/libCodec2JNI.so', missing and no known rule to make it`

BadDateAlex, 2024-06-14:
Have you ever import libcode2.so manually? I can't find it.

hemant-novotrax, 2024-07-01:
the same issue occurs..

sh123, 2024-07-13:
Hi, you need to use NDK 21/22 version, it is specified as 21.4.7075529 in build gradle

closed 2024-10-15 by sh123