sh123/codec2_talkie #81: Add Privacy option

Local42, 2025-06-06:
Could you add Privacy option as you did in esp32_loradv?

sh123, 2025-06-28:
While it is already supported, it is not compatible with the esp32_loradv project because esp32_loradv utilizes ChaCha20-Poly1305. In this app, you can enable privacy, but it employs AES 128 as an experimental feature without message authentication. This can be activated in Settings -> Set KISS basic parameters -> Enable KISS packet scrambling. Since this feature is experimental, I cannot provide any guarantees. If you require a private messaging solution, consider using another app, such as Meshastic or similar alternatives.

closed 2025-06-28 by sh123

Local42, 2025-06-28:
Could you give an example of the "Set key for speech scrambling". Thanks.

sh123, 2025-06-28:
You can set any arbitrary length master password, because it uses Android's `PBEKeySpec` with `PBEwithSHA256and128BITAES-CBC-BC` to derive secure AES 128 bit key, so you do not need to specify this key directly.

Local42, 2025-06-28:
Thanks a lot. For private messaging I use Reticulum.
https://reticulum.network/